sifiveinc/freedom-u-sdk #124: Centos build or prebuilt binaries

salmansheikh, 2019-09-11:
I can't get Freedom-u-sdk to build on Centos 7.  Are there any prebuilt binaries like there are for the freedom-e-sdk from SiFive https://www.sifive.com/boards

I get errors write away:
 make
make -C /home/ssheikh/freedom-u-sdk/buildroot RISCV=/home/ssheikh/freedom-u-sdk/work/buildroot_initramfs/host O=/home/ssheikh/freedom-u-sdk/work/buildroot_initramfs
make[1]: Entering directory `/home/ssheikh/freedom-u-sdk/buildroot'

You seem to have the current working directory in your
LD_LIBRARY_PATH environment variable. This doesn't work.
make[2]: *** [dependencies] Error 1
make[1]: *** [_all] Error 2
make[1]: Leaving directory `/home/ssheikh/freedom-u-sdk/buildroot'
rm -rf /home/ssheikh/freedom-u-sdk/work/riscv-pk
mkdir -p /home/ssheikh/freedom-u-sdk/work/riscv-pk
cd /home/ssheikh/freedom-u-sdk/work/riscv-pk && PATH=/home/ssheikh/freedom-u-sdk/work/buildroot_initramfs/host/bin:/home/ssheikh/anaconda3/bin:/home/ssheikh/anaconda3/condabin:/opt/Quartus/intelFPGA/18.1/quartus/bin:/usr/local/microsemi/Libero_SoC_v11.9/Libero/bin:/opt/Modelsim-2019.2/modeltech/bin:/opt/Quartus/intelFPGA/18.1/quartus/bin:/usr/local/microsemi/Libero_SoC_v11.9/Libero/bin:/opt/Modelsim-2019.2/modeltech/bin:/usr/lib64/qt-3.3/bin:/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin:/sbin:/home/ssheikh/.local/bin:/home/ssheikh/bin /home/ssheikh/freedom-u-sdk/riscv-pk/configure \
	--host=riscv64-sifive-linux-gnu \
	--enable-logo \
	--with-logo=/home/ssheikh/freedom-u-sdk/conf/sifive_logo.txt
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv64-sifive-linux-gnu
checking for riscv64-sifive-linux-gnu-gcc... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for riscv64-sifive-linux-gnu-g++... no
checking for riscv64-sifive-linux-gnu-c++... no
checking for riscv64-sifive-linux-gnu-gpp... no
checking for riscv64-sifive-linux-gnu-aCC... no
checking for riscv64-sifive-linux-gnu-CC... no
checking for riscv64-sifive-linux-gnu-cxx... no
checking for riscv64-sifive-linux-gnu-cc++... no
checking for riscv64-sifive-linux-gnu-cl.exe... no
checking for riscv64-sifive-linux-gnu-FCC... no
checking for riscv64-sifive-linux-gnu-KCC... no
checking for riscv64-sifive-linux-gnu-RCC... no
checking for riscv64-sifive-linux-gnu-xlC_r... no
checking for riscv64-sifive-linux-gnu-xlC... no
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking for riscv64-sifive-linux-gnu-ar... no
checking for ar... ar
checking for riscv64-sifive-linux-gnu-ranlib... no
checking for ranlib... ranlib
checking for riscv64-sifive-linux-gnu-readelf... no
checking for readelf... readelf
checking for riscv64-sifive-linux-gnu-objcopy... no
checking for objcopy... objcopy
checking for a BSD-compatible install... /usr/bin/install -c
checking how to run the C preprocessor... gcc -E
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for ANSI C header files... yes
configure: configuring default subproject : pk
configure: configuring default subproject : bbl
configure: configuring default subproject : softfloat
configure: configuring default subproject : dummy_payload
configure: configuring default subproject : machine
configure: configuring default subproject : util
configure: creating ./config.status
config.status: creating pk.mk
config.status: creating bbl.mk
config.status: creating softfloat.mk
config.status: creating dummy_payload.mk
config.status: creating machine.mk
config.status: creating util.mk
config.status: creating Makefile
config.status: creating config.h
CFLAGS="-mabi=lp64d -march=rv64imafdc" make PATH=/home/ssheikh/freedom-u-sdk/work/buildroot_initramfs/host/bin:/home/ssheikh/anaconda3/bin:/home/ssheikh/anaconda3/condabin:/opt/Quartus/intelFPGA/18.1/quartus/bin:/usr/local/microsemi/Libero_SoC_v11.9/Libero/bin:/opt/Modelsim-2019.2/modeltech/bin:/opt/Quartus/intelFPGA/18.1/quartus/bin:/usr/local/microsemi/Libero_SoC_v11.9/Libero/bin:/opt/Modelsim-2019.2/modeltech/bin:/usr/lib64/qt-3.3/bin:/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/bin:/sbin:/home/ssheikh/.local/bin:/home/ssheikh/bin -C /home/ssheikh/freedom-u-sdk/work/riscv-pk
make[1]: Entering directory `/home/ssheikh/freedom-u-sdk/work/riscv-pk'
gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE -mabi=lp64d -march=rv64imafdc   -DBBL_PAYLOAD=\"bbl_payload\" -DBBL_LOGO_FILE=\"bbl_logo_file\" -I. -I/home/ssheikh/freedom-u-sdk/riscv-pk/pk -I/home/ssheikh/freedom-u-sdk/riscv-pk/bbl -I/home/ssheikh/freedom-u-sdk/riscv-pk/softfloat -I/home/ssheikh/freedom-u-sdk/riscv-pk/dummy_payload -I/home/ssheikh/freedom-u-sdk/riscv-pk/machine -I/home/ssheikh/freedom-u-sdk/riscv-pk/util -c /home/ssheikh/freedom-u-sdk/riscv-pk/pk/file.c
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
gcc: error: unrecognized argument in option ‘-mabi=lp64d’
gcc: note: valid arguments to ‘-mabi=’ are: ms sysv
make[1]: *** [file.o] Error 1
make[1]: Leaving directory `/home/ssheikh/freedom-u-sdk/work/riscv-pk'
make: *** [/home/ssheikh/freedom-u-sdk/work/riscv-pk/bbl] Error 2

jim-wilson, 2019-09-11:
>You seem to have the current working directory in your
>LD_LIBRARY_PATH environment variable. This doesn't work.

Unset LD_LIBRARY_PATH like the error message tells you to do.

>gcc: error: unrecognized argument in option ‘-mcmodel=medany’
>gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
>gcc: error: unrecognized argument in option ‘-mabi=lp64d’

This indicates that you are trying to use an x86 compiler where a riscv compiler is required.  I think that setting the RISCV environment variable can cause this problem.  RISCV should not be set when building riscv-linux software.  it is only useful for embedded riscv software.

If you click on the releases tab in the github window, you can find some pre-built images, though that lack documentation on exactly how to use them.

You might also try using open-embedded, github.com/riscv/meta-riscv, we are planning to deprecate freedom-u-sdk and replace it with open embedded instead.

jim-wilson, 2019-09-13:
freedom-e-sdk is our BSP for bare metal programming.  We do have some support in there for the u540.  I see a qemu-sifive-u54 dir and a sifive-hifive-unleashed dir, though unfortunately I don't see a freedom-u540-vc707 dir, so I'm not sure if the other two will work for the vc707.  In theory, you can do bare metal programming on this, but there is a lot of stuff to learn to bring up an OS on a u540 part.

I haven't used a vc707 so I don't know the details of how to use it.

salmansheikh, 2019-09-17:
Okay, I got it to start compiling yesterday. Going to hopefully be able to
find some example programs and scripts (hopefully not from scratch) to
access some of the basic IO functions...want to put RTEMS on their if
possible.

Salman

tprrt, 2022-11-15:
This issue is no longer relevant.

closed 2022-11-15 by tprrt