sifiveinc/freedom-u-sdk #127: I want to load the device tree from the SD card

ImagineMiracle-wxn, 2019-10-18:
I need to remove the u500 device tree file from the free project and load it into the SD card by bbl. And already done so. My solution is this, write the device tree file and bbl to the SD card at the same time, and specify their location, 'freedom-u-sdk/riscv-pk/machine/minit.c' to load with this fixed address Device tree file. The device tree is not used in ‘freedom/bootrom/sdboot/head.S’. The bitstream and bbl image can be successfully generated, but when running on the u500, it will be stuck at the load bbl. That is to say, the print message "BOOT" stops working.
[image]

ImagineMiracle-wxn, 2019-10-18:
And, What is the role of 0x7ffff000?
'freedom/bootrom/sdbot/head.S'
[image]

jim-wilson, 2019-10-18:
If you are modifying sources and running into problems, then you will likely have to debug them yourself.  freedom-u-sdk isn't a very good place to ask questions about boot loading and hardware.  This is just a linux distro.  The 0x7fff000 looks like an arbitrary offset for the stack pointer, far enough away that it won't conflict with what is being loaded into memory, but not so far that it runs off the end of physical memory or causes addressing problems.

ImagineMiracle-wxn, 2019-10-23:
Thanks! What is the role of "format-boot-loader" in the Makefile file?

jim-wilson, 2019-10-23:
format-boot-loader can create a bootable SDcard for the Hifive Unleashed.

ImagineMiracle-wxn, 2019-10-25:
So,Can't it be used on u500-fpga?

tmagik, 2019-10-25:
Currently no. Several things need to be updated/develop for that to work

closed 2022-11-15 by tprrt