sifiveinc/freedom-u-sdk #130: Will U500 be a target soon?

bchetwynd, 2019-11-26:
Are there plans porting the Yocto-based freedom-u-sdk to the Freedom U500 platform?

Or... can guidance be provided about how to creating a U500 machine target?

Thanks,
Brendon

tmagik, 2019-11-28:
My suggestion is start with this:

https://github.com/sifive/freedom-u540-c000-bootloader

to replace the FPGA ‘rom’ with the same boot flow we use on U540. You’ll want to
remove and/or #ifdef out anything related to DDR init, PLLs, etc, as all of that is 
set up by the FPGA bitstream.

My goal is that the exact same SDcard image we build with Yocto/OE should
‘just work’ on an FPGA U500 machine, provided we are passing in the right 
device tree. (Really, the ROM/zsbl bootloader should probably be a subproject
of sifive/freedom and then rocket-chip needs a few updates to generate device
trees that are compatible with upstream linux.)

tprrt, 2022-11-15:
This issue is no longer relevant.

closed 2022-11-15 by tprrt