sifiveinc/freedom-u-sdk #133: ERROR: Layer meta-sifive is not compatible with the core layer which only supports these series: zeus (layer is compatible with warrior)

weiqi7777, 2020-02-09:
I try to build linux images according to the guide.

but i get below error
```bash
lujun@softserver build $ MACHINE=freedom-u540 bitbake demo-coreip-cli
ERROR: Unable to start bitbake server (None)
ERROR: Server log for this session (/home/lujun/work/sifive/riscv-sifive/build/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 15407 at 2020-02-09 12:52:28.027779 ---
ERROR: Layer meta-sifive is not compatible with the core layer which only supports these series: zeus (layer is compatible with warrior)
```

the complete console info is below:
```bash
lujun@softserver riscv-sifive $ source meta-sifive/setup.sh 
Init OE
You had no conf/local.conf file. This configuration file has therefore been
created for you with some default values. You may wish to edit it to, for
example, select a different MACHINE (target hardware). See conf/local.conf
for more information as common configuration options are commented.

You had no conf/bblayers.conf file. This configuration file has therefore been
created for you with some default values. To add additional metadata layers
into your configuration please add entries to conf/bblayers.conf.

The Yocto Project has extensive documentation about OE including a reference
manual which can be found at:
    http://yoctoproject.org/documentation

For more information about OpenEmbedded see their website:
    http://www.openembedded.org/

### Shell environment set up for builds. ###

You can now run 'bitbake <target>'

Common targets are:
    core-image-minimal
    core-image-sato
    meta-toolchain
    meta-ide-support

You can also run generated qemu images with a command like 'runqemu qemux86'.

Other commonly useful commands are:
 - 'devtool' and 'recipetool' handle common recipe tasks
 - 'bitbake-layers' handles common layer tasks
 - 'oe-pkgdata-util' handles common target package tasks
Adding layers
NOTE: Starting bitbake server...
NOTE: Starting bitbake server...
g NOTE: Starting bitbake server...
NOTE: Starting bitbake server...
me	NOTE: Starting bitbake server...
NOTE: Starting bitbake server...
NOTE: Starting bitbake server...
NOTE: Starting bitbake server...
Creating auto.conf
---------------------------------------------------
MACHINE=freedom-u540 bitbake demo-coreip-cli
---------------------------------------------------

Buildable machine info
---------------------------------------------------
* freedom-u540: The SiFive HiFive Unleased board
* qemuriscv64: The 64-bit RISC-V machine
---------------------------------------------------
lujun@softserver build $ g ls
lujun@softserver build $ MACHINE=freedom-u540 bitbake demo-coreip-cli
ERROR: Unable to start bitbake server (None)
ERROR: Server log for this session (/home/lujun/work/sifive/riscv-sifive/build/bitbake-cookerdaemon.log):
--- Starting bitbake server pid 15407 at 2020-02-09 12:52:28.027779 ---
ERROR: Layer meta-sifive is not compatible with the core layer which only supports these series: zeus (layer is compatible with warrior)
```

can you help me to solve the problem?

davidlt, 2020-02-09:
Could you share full commands you executed? I have built it two days ago successfully. The message would indicate that layers are mismatched from what's expected in the manifest ( https://github.com/sifive/meta-sifive/blob/v201908-branch/tools/manifests/sifive.xml ).

cck-wcs, 2020-02-11:
I got the same problem too. The xml file is the same as you list above.

cck-wcs, 2020-02-11:
> I got the same problem too. The xml file is the same as you list above.

OK, my issue is fixed. Maybe I forgot to execute the command 'repo sync'. Thanks.

tprrt, 2022-11-15:
This issue is no longer relevant.

closed 2022-11-15 by tprrt