sifiveinc/freedom-u-sdk #134: Build failed with error

Sayeed42, 2020-02-21:
Hi,
I tried to build the image according to the Readme file. But the build failed. Can you please help?
The log from the console is posted here, as I could not attach the log file.

`MACHINE=freedom-u540 bitbake demo-coreip-cli
Parsing recipes: 100% |###############################################################################################################################################################################################| Time: 0:02:01
Parsing of 2252 .bb files complete (0 cached, 2252 parsed). 3271 targets, 154 skipped, 47 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.04"
TARGET_SYS           = "riscv64-oe-linux"
MACHINE              = "freedom-u540"
DISTRO               = "nodistro"
DISTRO_VERSION       = "nodistro.0"
TUNE_FEATURES        = "riscv64"
meta                 = "work:ffa91ff19f73c3ed4060a6ecd3b6da7f46732cb3"
meta-oe              
meta-python          
meta-multimedia      
meta-networking      
meta-gnome           
meta-xfce            = "work:696408ce3661b663a7895f75a01613e6cfeb8186"
meta-riscv           = "work:590357c49bf85e2eadf4068e53d2511b468737cc"
meta-sifive          = "work:23a8b1b20a6a929b3f15625b0d4b8fabbd1362dc"

NOTE: Fetching uninative binary shim from http://downloads.yoctoproject.org/releases/uninative/2.7/x86_64-nativesdk-libc.tar.xz;sha256sum=9498d8bba047499999a7310ac2576d0796461184965351a56f6d32c888a1f216
Initialising tasks: 100% |############################################################################################################################################################################################| Time: 0:00:05
Sstate summary: Wanted 1983 Found 0 Missed 1983 Current 0 (0% match, 0% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
WARNING: less-551-r0 do_fetch: Failed to fetch URL http://www.greenwoodsoftware.com/less/less-551.tar.gz, attempting MIRRORS if available
ERROR: perf-1.0-r9 do_compile: oe_runmake failed
ERROR: perf-1.0-r9 do_compile: Execution of '/home/sayeed42/riscv-sifive/build/tmp-glibc/work/freedom_u540-oe-linux/perf/1.0-r9/temp/run.do_compile.31084' failed with exit code 1:
make: Entering directory '/home/sayeed42/riscv-sifive/build/tmp-glibc/work/freedom_u540-oe-linux/perf/1.0-r9/perf-1.0/tools/perf'
  BUILD:   Doing 'make -j4' parallel build
  HOSTCC   /home/sayeed42/riscv-sifive/build/tmp-glibc/work/freedom_u540-oe-linux/perf/1.0-r9/perf-1.0/fixdep.o
  HOSTLD   /home/sayeed42/riscv-sifive/build/tmp-glibc/work/freedom_u540-oe-linux/perf/1.0-r9/perf-1.0/fixdep-in.o
  LINK     /home/sayeed42/riscv-sifive/build/tmp-glibc/work/freedom_u540-oe-linux/perf/1.0-r9/perf-1.0/fixdep

Auto-detecting system features:
...                         dwarf: [ on  ]
...            dwarf_getlocations: [ on  ]
...                         glibc: [ on  ]
...                          gtk2: [ OFF ]
...                      libaudit: [ OFF ]
...                        libbfd: [ on  ]
...                        libcap: [ OFF ]
...                        libelf: [ on  ]
...                       libnuma: [ OFF ]
...        numa_num_possible_cpus: [ OFF ]
...                       libperl: [ OFF ]
...                     libpython: [ OFF ]
...                     libcrypto: [ on  ]
...                     libunwind: [ OFF ]
...            libdw-dwarf-unwind: [ on  ]
...                          zlib: [ on  ]
...                          lzma: [ on  ]
...                     get_cpuid: [ OFF ]
...                           bpf: [ on  ]
...                        libaio: [ on  ]
...                       libzstd: [ OFF ]
...        disassembler-four-args: [ on  ]

Makefile.config:506: DWARF support is off, BPF prologue is disabled
Makefile.config:589: Disabling post unwind, no support found.
Makefile.config:714: Missing perl devel files. Disabling perl scripting support, please install perl-ExtUtils-Embed/libperl-dev
Makefile.config:750: No 'Python.h' (for Python 2.x support) was found: disables Python support - please install python-devel/python-dev
[... 416 lines cut ...]
|   CC       /home/sayeed42/riscv-sifive/build/tmp-glibc/work/freedom_u540-oe-linux/perf/1.0-r9/perf-1.0/tep_strerror.o
|   CC       /home/sayeed42/riscv-sifive/build/tmp-glibc/work/freedom_u540-oe-linux/perf/1.0-r9/perf-1.0/event-parse-api.o
|   LD       /home/sayeed42/riscv-sifive/build/tmp-glibc/work/freedom_u540-oe-linux/perf/1.0-r9/perf-1.0/libtraceevent-in.o
|   LINK     /home/sayeed42/riscv-sifive/build/tmp-glibc/work/freedom_u540-oe-linux/perf/1.0-r9/perf-1.0/libtraceevent.a
| Makefile.perf:224: recipe for target 'sub-make' failed
| make[1]: *** [sub-make] Error 2
| Makefile:69: recipe for target 'all' failed
| make: *** [all] Error 2
| make: Leaving directory '/home/sayeed42/riscv-sifive/build/tmp-glibc/work/freedom_u540-oe-linux/perf/1.0-r9/perf-1.0/tools/perf'
| WARNING: /home/sayeed42/riscv-sifive/build/tmp-glibc/work/freedom_u540-oe-linux/perf/1.0-r9/temp/run.do_compile.31084:1 exit 1 from 'exit 1'
| 
ERROR: Task (/home/sayeed42/riscv-sifive/openembedded-core/meta/recipes-kernel/perf/perf.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3537 tasks of which 0 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory
NOTE: Writing buildhistory took: 9 seconds
NOTE: Build completion summary:
NOTE:   do_populate_sysroot: 0.0% sstate reuse(0 setscene, 275 scratch)
NOTE:   do_package: 0.0% sstate reuse(0 setscene, 197 scratch)
NOTE:   do_packagedata: 0.0% sstate reuse(0 setscene, 183 scratch)
NOTE:   do_package_write_ipk: 0.0% sstate reuse(0 setscene, 97 scratch)

Summary: 1 task failed:
  /home/sayeed42/riscv-sifive/openembedded-core/meta/recipes-kernel/perf/perf.bb:do_compile
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.`

jim-wilson, 2020-02-22:
Failed to import the site module
Traceback (most recent call last):
File "/usr/lib/python3.6/site.py", line 570, in
main()
...
 File "/usr/lib/python3.6/sysconfig.py", line 421, in _init_posix
| _temp = import(name, globals(), locals(), ['build_time_vars'], 0)
| ModuleNotFoundError: No module named '_sysconfigdata'

This looks like a problem with your local python3 install.  Or maybe an environment variable problem.  I have 18.04.04 on my laptop and I can build open embedded on it.

Sayeed42, 2020-02-23:
I re-installed python3-distutils and tried to build again. Then I got a new error! Please help!
`ERROR: nss-3.50-r0 do_compile: Execution of '/home/sayeed42/riscv-sifive/build/tmp-glibc/work/riscv64-oe-linux/nss/3.50-r0/temp/run.do_compile.461' failed with exit code 2`

jim-wilson, 2020-02-23:
I have nss-3.45 in my source tree.  How did you get nss-3.50?  I think you have a bleeding edge copy of openembedded, and are trying to build stuff that we haven't tested yet, and which may not actually build.  If I go into openembedded-core and do "git checkout master" then I see nss-3.50 in my source tree.  But that is definitely something you should not do unless you know how to debug these kinds of build failures, and want to contribute to openembedded.  We can only support the version of openembedded that we are building and testing.  If I do a "repo sync" in the riscv-sifive dir, then I get back to the supported copy of openembedded-core which has nss-3.45.

I'm not a repo expert, so I don't know if there is a good way to figure out what you have.  Maybe compare "repo info" output?

Sayeed42, 2020-03-25:
As it turned out, if I start building the image after executing "repo init" and "repo sync", the build completes successfully. Previously, I was also executing "repo rebase" after "repo sync", which was pulling the untested components, which in turn, was causing the build to fail. Thanks, @jim-wilson, for helping me out. I am closing this issue.

closed 2020-03-25 by Sayeed42