sifiveinc/freedom-u-sdk #137: What happened to this repo?

aswaterman, 2020-04-22:
It seems like https://github.com/sifive/freedom-u-sdk/tree/archive/buildroot is more useful than the current master branch--at least from my perspective as a RISC-V developer.  What's with the lame new version?

knightsifive, 2020-04-22:
I've been happily using the new Yocto/OE FUSDK. I find it friendlier than the old Buildroot version. I'm sure the developers would appreciate feature requests and other constructive feedback.

davidlt, 2020-04-22:
Freedom U SDK moved from using buildroot to OpenEmbedded (OE) [sometimes people also refer to it using Yocto name] to build disk images. That came with a number of changes (incl. latest kernels and other packages, bootflow, SBI implementation [from BBL to OpenSBI]). These changes to require some effort to adapt for your particular work. The current targets we have in the new FUSDK might not yet provide what was available in buildroot based images, but hopefully we will improve that in the future.

buildroot branch is still available on GitHub if there is a need: https://github.com/sifive/freedom-u-sdk/tree/archive/buildroot but is not supported.

closed 2020-04-22 by davidlt