sifiveinc/freedom-u-sdk #141: UEFI?

baryluk, 2020-12-08:
Are there plans to support UEFI + ACPI on HiFive Unmatched for example? If not, why not?

I hate device tree files with passion (it is a huge mess and pain on ARM).

Something like ARM SBSA for RISC-V would be really super useful for running Linux out of the box.

Booting from USB and from Network (TFTP/PXE) also would be a plus, but are orthogonal to UEFI and ACPI.

davidlt, 2020-12-09:
UEFI is enabled by default in U-Boot for RISC-V (or at least existing boards). No one is working on ACPI, which is not really needed. Even on Unleashed (not Unmatched) one could boot with TFTP and PXE with U-Boot.

There is also a Tianocore EDK2 port for Unleashed (not Unmathced) by HPE which is upstreamed. So in general Unleashed board has two UEFI implementations working: U-Boot and Tianocore EDK2. Both were tested to boot Linux.

SiFive specification are here (WIP): RISC-V Profiles and Platform Specification, https://github.com/riscv/riscv-platform-specs
Also see: https://github.com/riscv/riscv-sbi-doc

Realistically if DTB is embedded into FW general users should see no different between ACPI vs DTB.

Community have showed that we can do: --> U-Boot --(EFI) ---> GRUB2 [EFI] --(EFI)--> Kernel (EFI).

EFI Stubs were merged in v5.10 which is not yet released. GRUB2 patches were posted, needs rework to support LoadFifle2 (which ARM recently posted, not in arch-generic way, thus again rework needed).

baryluk, 2020-12-09:
That is really awesome to hear. Thank you for detailed answer and pointers to the software.

closed 2020-12-09 by baryluk

baryluk, 2020-12-09:
Just one more question. Where would DTB reside on these existing boards? On a soldered flash chip on board?

davidlt, 2020-12-09:
DTB can be in various places (depends how you like to build your bootflow): ZSBL, FSBL, OpenSBI, U-Boot and external (e.g. in /boot partition).

ZSBL is part of SoC ROM, thus that one cannot be modified and it's simply ignored by the next boot stage. After that all options are open. Each Linux distribution (e.g. Fedora, Debian, Gentoo, SUSE, Ubuntu, etc.)  might do a different thing. Unless there is some restrictions based on RISC-V Profiles and Platforms specification.

The current bootflow of FUSDK loads DTB from /boot partition.

baryluk, 2020-12-09:
What does it mean "can"? If I receive the board from SiFive, where is it?

davidlt, 2020-12-09:
That means that developers who design bootflow can put DTB  in many places.

The current bootflow of FUSDK loads DTB from /boot partition, i.e. it's not embedded into SPI NAND Flash. This allows developers to modify DTB (or use a different one) without recompiling, embedding, and flashing.