sifiveinc/freedom-u-sdk #142: How can I support python on Xilinx Vivado FPGA board?

nagmat1, 2021-02-17:
I just installed freedom-u-sdk on Xilinx Vivado FPGA board according to these steps:
```git clone https://github.com/mcd500/freedom-u-sdk
cd freedom-u-sdk
git checkout -b linux_u500vc707devkit_config origin/linux_u500vc707devkit_config
git submodule update --init --recursive --force
make -j4``` 
Then I wrote a binary and a tool which worked on freedom-u-sdk simulator, but when I tried it on Hardware, python was not working on it. Now my question is: How can I support python on Xilinx Vivado Hardware?

jim-wilson, 2021-02-17:
This is an old pre-upstream buildroot based system which we don't support anymore as far as I know.  You should be able to modify the buildroot config to enable python support.  Whether that actually builds and works, I have no idea.  You would have to try it and see.

nagmat1, 2021-02-18:
Thank for your quick response! Yeah, it works fine on my board. What are the steps to enable python support? Could you share some resource please?

jim-wilson, 2021-02-18:
You can find buildoot docs here.
https://buildroot.org/downloads/manual/manual.html
You want the user guide section which talks about how to configure buildroot.

closed 2021-02-19 by nagmat1