sifiveinc/freedom-u-sdk #145: task failed regarding to bb:do_compile

yslys, 2021-05-24:
Hi, I am currently following the instructions to set up the environment. However, errors occurred when I was building the images on Ubuntu 20.04.2 LTS, with Linux kernel version 5.8.0-53-generic.

Below is the part of the output when I build:
```
| [2667/5158] /home/yusen/Desktop/riscv-sifive/build/tmp-glibc/hosttools/g++ -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -Itools/clang/lib/Sema -I/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/clang/lib/Sema -I/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/clang/include -Itools/clang/include -Iinclude -I/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/llvm/include -isystem/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work/x86_64-linux/clang-native/11.0.1-r0/recipe-sysroot-native/usr/include -O2 -pipe   -fPIC -fvisibility-inlines-hidden -Werror=date-time -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -pedantic -Wno-long-long -Wimplicit-fallthrough -Wno-maybe-uninitialized -Wno-class-memaccess -Wno-redundant-move -Wno-noexcept-type -Wdelete-non-virtual-dtor -Wno-comment -fdiagnostics-color -ffunction-sections -fdata-sections -fno-common -Woverloaded-virtual -fno-strict-aliasing -isystem/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work/x86_64-linux/clang-native/11.0.1-r0/recipe-sysroot-native/usr/include -O2 -pipe   -DNDEBUG -g0 -std=c++14 -MD -MT tools/clang/lib/Sema/CMakeFiles/obj.clangSema.dir/SemaExpr.cpp.o -MF tools/clang/lib/Sema/CMakeFiles/obj.clangSema.dir/SemaExpr.cpp.o.d -o tools/clang/lib/Sema/CMakeFiles/obj.clangSema.dir/SemaExpr.cpp.o -c /home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/clang/lib/Sema/SemaExpr.cpp
| FAILED: tools/clang/lib/Sema/CMakeFiles/obj.clangSema.dir/SemaExpr.cpp.o
| /home/yusen/Desktop/riscv-sifive/build/tmp-glibc/hosttools/g++ -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -Itools/clang/lib/Sema -I/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/clang/lib/Sema -I/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/clang/include -Itools/clang/include -Iinclude -I/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/llvm/include -isystem/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work/x86_64-linux/clang-native/11.0.1-r0/recipe-sysroot-native/usr/include -O2 -pipe   -fPIC -fvisibility-inlines-hidden -Werror=date-time -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -pedantic -Wno-long-long -Wimplicit-fallthrough -Wno-maybe-uninitialized -Wno-class-memaccess -Wno-redundant-move -Wno-noexcept-type -Wdelete-non-virtual-dtor -Wno-comment -fdiagnostics-color -ffunction-sections -fdata-sections -fno-common -Woverloaded-virtual -fno-strict-aliasing -isystem/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work/x86_64-linux/clang-native/11.0.1-r0/recipe-sysroot-native/usr/include -O2 -pipe   -DNDEBUG -g0 -std=c++14 -MD -MT tools/clang/lib/Sema/CMakeFiles/obj.clangSema.dir/SemaExpr.cpp.o -MF tools/clang/lib/Sema/CMakeFiles/obj.clangSema.dir/SemaExpr.cpp.o.d -o tools/clang/lib/Sema/CMakeFiles/obj.clangSema.dir/SemaExpr.cpp.o -c /home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/clang/lib/Sema/SemaExpr.cpp
| {standard input}: Assembler messages:
| {standard input}:138929: Warning: end of file not at end of a line; newline inserted
| {standard input}: Error: open CFI at the end of file; missing .cfi_endproc directive
| g++: fatal error: Killed signal terminated program cc1plus
| compilation terminated.
| [2668/5158] /home/yusen/Desktop/riscv-sifive/build/tmp-glibc/hosttools/g++ -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -Itools/clang/lib/Sema -I/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/clang/lib/Sema -I/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/clang/include -Itools/clang/include -Iinclude -I/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/llvm/include -isystem/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work/x86_64-linux/clang-native/11.0.1-r0/recipe-sysroot-native/usr/include -O2 -pipe   -fPIC -fvisibility-inlines-hidden -Werror=date-time -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -pedantic -Wno-long-long -Wimplicit-fallthrough -Wno-maybe-uninitialized -Wno-class-memaccess -Wno-redundant-move -Wno-noexcept-type -Wdelete-non-virtual-dtor -Wno-comment -fdiagnostics-color -ffunction-sections -fdata-sections -fno-common -Woverloaded-virtual -fno-strict-aliasing -isystem/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work/x86_64-linux/clang-native/11.0.1-r0/recipe-sysroot-native/usr/include -O2 -pipe   -DNDEBUG -g0 -std=c++14 -MD -MT tools/clang/lib/Sema/CMakeFiles/obj.clangSema.dir/SemaTemplateInstantiate.cpp.o -MF tools/clang/lib/Sema/CMakeFiles/obj.clangSema.dir/SemaTemplateInstantiate.cpp.o.d -o tools/clang/lib/Sema/CMakeFiles/obj.clangSema.dir/SemaTemplateInstantiate.cpp.o -c /home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/clang/lib/Sema/SemaTemplateInstantiate.cpp
| [2669/5158] /home/yusen/Desktop/riscv-sifive/build/tmp-glibc/hosttools/g++ -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -Itools/clang/lib/Sema -I/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/clang/lib/Sema -I/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/clang/include -Itools/clang/include -Iinclude -I/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/llvm/include -isystem/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work/x86_64-linux/clang-native/11.0.1-r0/recipe-sysroot-native/usr/include -O2 -pipe   -fPIC -fvisibility-inlines-hidden -Werror=date-time -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -pedantic -Wno-long-long -Wimplicit-fallthrough -Wno-maybe-uninitialized -Wno-class-memaccess -Wno-redundant-move -Wno-noexcept-type -Wdelete-non-virtual-dtor -Wno-comment -fdiagnostics-color -ffunction-sections -fdata-sections -fno-common -Woverloaded-virtual -fno-strict-aliasing -isystem/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work/x86_64-linux/clang-native/11.0.1-r0/recipe-sysroot-native/usr/include -O2 -pipe   -DNDEBUG -g0 -std=c++14 -MD -MT tools/clang/lib/Sema/CMakeFiles/obj.clangSema.dir/SemaTemplateDeduction.cpp.o -MF tools/clang/lib/Sema/CMakeFiles/obj.clangSema.dir/SemaTemplateDeduction.cpp.o.d -o tools/clang/lib/Sema/CMakeFiles/obj.clangSema.dir/SemaTemplateDeduction.cpp.o -c /home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/clang/lib/Sema/SemaTemplateDeduction.cpp
| [2670/5158] /home/yusen/Desktop/riscv-sifive/build/tmp-glibc/hosttools/g++ -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -Itools/clang/lib/Sema -I/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/clang/lib/Sema -I/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/clang/include -Itools/clang/include -Iinclude -I/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work-shared/llvm-project-source-11.0.1-r0/git/llvm/include -isystem/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work/x86_64-linux/clang-native/11.0.1-r0/recipe-sysroot-native/usr/include -O2 -pipe   -fPIC -fvisibility-inlines-hidden -Werror=date-time -Wall -Wextra -Wno-unused-parameter -Wwrite-strings -Wcast-qual -Wno-missing-field-initializers -pedantic -Wno-long-long -Wimplicit-fallthrough -Wno-maybe-uninitialized -Wno-class-memaccess -Wno-redundant-move -Wno-noexcept-type -Wdelete-non-virtual-dtor -Wno-comment -fdiagnostics-color -ffunction-sections -fdata-sections -fno-common -Woverloaded-virtual -fno-strict-aliasing -isystem/home/yusen/Desktop/riscv-sifive/build/tmp-glibc/work/x86_64-linux/clang-native/11.0.1-r0/recipe-sysroot-native/usr/include -O2 -p [... cut ...]
NOTE:   do_package: 0.0% sstate reuse(0 setscene, 88 scratch)
NOTE:   do_packagedata: 0.0% sstate reuse(0 setscene, 87 scratch)

Summary: 2 tasks failed:
  /home/yusen/Desktop/riscv-sifive/openembedded-core/meta/recipes-support/gnutls/gnutls_3.7.1.bb:do_compile
  virtual:native:/home/yusen/Desktop/riscv-sifive/meta-clang/recipes-devtools/clang/clang_git.bb:do_compile
Summary: There were 2 WARNING messages shown.
Summary: There were 3 ERROR messages shown, returning a non-zero exit code.
```

Could anyone please help me solve this? Thanks in advance!

davidlt, 2021-05-24:
Try using OE prebuilt build tools (see README for more details):
```
./openembedded-core/scripts/install-buildtools -r yocto-3.2_M2 -t 20200729
```
But looking again at it `g++: fatal error: Killed signal terminated program cc1plus`. I assume out-of-memory kill happened here due to lack of RAM. Adjust `PARALLEL_MAKE=` `BB_NUMBER_THREADS=` according to your machine. You can also build Clang first before building the full disk image.

yslys, 2021-05-24:
Thanks a lot @davidlt , I agree that by removing those two parameter settings would it work, as I have successfully built the disk image following this [link](https://github.com/riscv/meta-riscv) without specifying ```PARALLEL_MAKE``` and ```BB_NUMBER_THREADS```. I would close this issue after I have done the building images stuff.

davidlt, 2021-05-24:
Instructions are similar bu the content is different. LLVM/Clang stack is expensive resources wise to build.

yslys, 2021-05-29:
Sorry for the late reply. The building process ate up all my disk memory (up to 100 Gbs on my virtual machine) - it failed. May I know how much disk memory is required to build this?

yslys, 2021-05-30:
I retried using a virtual machine with 300 GBs of disk space and it built successfully, with 130 GB used.

Will close this issue.

closed 2021-05-30 by yslys