sifiveinc/freedom-u-sdk #146: Out of memory when running tasks on riscv64-linux on qemu

yslys, 2021-05-31:
Hi all,

I have successfully set up the environment running on qemu with ```demo-coreip-cli``` - the basic command line image.

I have a project that wants to build inside the environment, but the build process is always killed due to out of memory. I executed the ```free``` command, and it shows the following:
```
root@qemuriscv64:~/hmc/src/build# free
                       total         used         free          shared    buff/cache   available
Mem:           169372       48192       68756       24952       52424       89644
Swap:              0           0           0

```

May I know if there is a way to configure the memory to make it larger when executing the ```MACHINE=qemuriscv64 bitbake demo-coreip-cli``` command? Thanks in advance.

jim-wilson, 2021-06-02:
You can pass qemu args to runqemu to change the memory allocated.
    https://docs.yoctoproject.org/dev-manual/qemu.html
and see section 4.9.  So adding qemuparams="-m 1024" will give you 1G of memory.  -m 256 is the default.

The default is set in openembedded-code/meta/classes/qemuboot.bbclass on the QB_MEM line.   You can change it there, rerun bitbake, and then you don't need to pass extra arguments.  Depending on what you are building, it is possible that QB_MEM might be overridden in other files, some image files override it for instance.

yslys, 2021-06-02:
Thanks Jim, your solution perfectly solves the issue, and I would close this issue.

closed 2021-06-02 by yslys