sifiveinc/freedom-u-sdk #147: Building error: libpmem.h not found

yslys, 2021-06-06:
Hi all,
I tried to build following the instructions but it showed that ```libpmem.h``` not found. My system is Ubuntu 18.04, with kernel version 4.15.0-142-generic.

Part of the error message is shown below:

```
| [531/2467] Compiling C object libqemuutil.a.p/util_qht.c.o
| [532/2467] Compiling C object libqemuutil.a.p/util_guest-random.c.o
| [533/2467] Compiling C object libqemuutil.a.p/util_aio-wait.c.o
| [534/2467] Compiling C object libcommon.fa.p/hw_isa_apm.c.o
| [535/2467] Compiling C object libcommon.fa.p/hw_mem_pc-dimm.c.o
| [536/2467] Compiling C object libcommon.fa.p/hw_isa_piix3.c.o
| [537/2467] Compiling C object libcommon.fa.p/hw_mem_nvdimm.c.o
| FAILED: libcommon.fa.p/hw_mem_nvdimm.c.o
| gcc -Ilibcommon.fa.p -I. -I../qemu-5.2.0 -I../qemu-5.2.0/capstone/include/capstone -I../qemu-5.2.0/dtc/libfdt -I../qemu-5.2.0/slirp -I../qemu-5.2.0/slirp/src -Iqapi -Itrace -Iui -Iui/shader -I/nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/include/pixman-1 -I/nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/include -I/nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/include/SDL2 -I/nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/include/libmount -I/nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/include/blkid -I/nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/include/glib-2.0 -I/nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/lib/glib-2.0/include -I/nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/include/gio-unix-2.0 -I/nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/include/virgl -I/nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/include/libdrm -I/nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/recipe-sysroot-native/usr/share/pkgconfig/../../../usr/include -fdiagnostics-color=auto -pipe -Wall -Winvalid-pch -std=gnu99 -O2 -g -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=2 -m64 -mcx16 -D_GNU_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -Wstrict-prototypes -Wredundant-decls -Wundef -Wwrite-strings -Wmissing-prototypes -fno-strict-aliasing -fno-common -fwrapv -O2 -Wold-style-declaration -Wold-style-definition -Wtype-limits -Wformat-security -Wformat-y2k -Winit-self -Wignored-qualifiers -Wempty-body -Wnested-externs -Wendif-labels -Wexpansion-to-defined -Wno-missing-include-dirs -Wno-shift-negative-value -Wno-psabi -fstack-protector-strong -isystem /nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/qemu-5.2.0/linux-headers -isystem linux-headers -iquote /nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/qemu-5.2.0/tcg/i386 -iquote . -iquote /nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/qemu-5.2.0 -iquote /nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/qemu-5.2.0/accel/tcg -iquote /nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/qemu-5.2.0/include -iquote /nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/qemu-5.2.0/disas/libvixl -pthread -isystem/nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/recipe-sysroot-native/usr/include -O2 -D_REENTRANT -Wno-undef -MD -MQ libcommon.fa.p/hw_mem_nvdimm.c.o -MF libcommon.fa.p/hw_mem_nvdimm.c.o.d -o libcommon.fa.p/hw_mem_nvdimm.c.o -c ../qemu-5.2.0/hw/mem/nvdimm.c
| In file included from ../qemu-5.2.0/hw/mem/nvdimm.c:27:
| /nobackup/yusen/sifive-U-SDK/build/tmp-glibc/work/x86_64-linux/qemu-system-native/5.2.0-r0/qemu-5.2.0/include/qemu/pmem.h:16:10: fatal error: libpmem.h: No such file or directory
|    16 | #include <libpmem.h>
|       |          ^~~~~~~~~~~
| compilation terminated.
```

I checked the packages installed using ```apt list | grep pmem```. The corresponding output is:
```
libpmem-dev/bionic-updates,bionic-updates,now 1.4.1-0ubuntu1~18.04.1 amd64 [installed]
libpmem1/bionic-updates,bionic-updates,now 1.4.1-0ubuntu1~18.04.1 amd64 [installed,automatic]
libpmem1-debug/bionic-updates,bionic-updates,now 1.4.1-0ubuntu1~18.04.1 amd64 [installed]
libpmemblk-dev/bionic-updates,bionic-updates 1.4.1-0ubuntu1~18.04.1 amd64
libpmemblk1/bionic-updates,bionic-updates,now 1.4.1-0ubuntu1~18.04.1 amd64 [installed,automatic]
libpmemblk1-debug/bionic-updates,bionic-updates 1.4.1-0ubuntu1~18.04.1 amd64
libpmemcto-dev/bionic-updates,bionic-updates 1.4.1-0ubuntu1~18.04.1 amd64
libpmemcto1/bionic-updates,bionic-updates,now 1.4.1-0ubuntu1~18.04.1 amd64 [installed,automatic]
libpmemcto1-debug/bionic-updates,bionic-updates 1.4.1-0ubuntu1~18.04.1 amd64
libpmemlog-dev/bionic-updates,bionic-updates,now 1.4.1-0ubuntu1~18.04.1 amd64 [installed]
libpmemlog1/bionic-updates,bionic-updates,now 1.4.1-0ubuntu1~18.04.1 amd64 [installed,automatic]
libpmemlog1-debug/bionic-updates,bionic-updates 1.4.1-0ubuntu1~18.04.1 amd64
libpmemobj-dev/bionic-updates,bionic-updates,now 1.4.1-0ubuntu1~18.04.1 amd64 [installed]
libpmemobj-doc/bionic-updates,bionic-updates,bionic-updates,bionic-updates,now 1.4.1-0ubuntu1~18.04.1 all [installed]
libpmemobj1/bionic-updates,bionic-updates,now 1.4.1-0ubuntu1~18.04.1 amd64 [installed,automatic]
libpmemobj1-debug/bionic-updates,bionic-updates 1.4.1-0ubuntu1~18.04.1 amd64
libpmempool-dev/bionic-updates,bionic-updates,now 1.4.1-0ubuntu1~18.04.1 amd64 [installed]
libpmempool1/bionic-updates,bionic-updates,now 1.4.1-0ubuntu1~18.04.1 amd64 [installed,automatic]
libpmempool1-debug/bionic-updates,bionic-updates 1.4.1-0ubuntu1~18.04.1 amd64
librpmem-dev/bionic-updates,bionic-updates 1.4.1-0ubuntu1~18.04.1 amd64
librpmem1/bionic-updates,bionic-updates 1.4.1-0ubuntu1~18.04.1 amd64
librpmem1-debug/bionic-updates,bionic-updates 1.4.1-0ubuntu1~18.04.1 amd64
```

I am not sure why this error occurs. May I know if it is because I have some packages not installed? Thanks in advance.

jim-wilson, 2021-06-06:
"apt list" shows all packages.  "apt list --installed" shows only installed packages.  I would expect libpmem-dev to contain the missing header file.  qemu configure uses "pkg-config --exists libpmem" to check if pmem is available, and if so tries to use it.  pkg-config libpmem returns 1 (no) on my system.  You perhaps have some uncommon package installed that makes the pkg-config libpmem check succeed, and then the build requires the header file.

yslys, 2021-06-06:
Thanks a lot for your reply. I execute ```pkg-config --exists libpmem```, and there is no output; I then execute with the ```print-errors``` flag on: ```pkg-config --exists libpmem --print-errors```, there is no return either. May I know if that means the pkg-config libpmem check succeeds?

If I execute ```pkg-config --exists libpmem-dev --print-errors```, the output is:
```
Package libpmem-dev was not found in the pkg-config search path.
Perhaps you should add the directory containing `libpmem-dev.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libpmem-dev' found
```
However, when I checked the installed packages by executing ```apt list --installed | grep libpmem-dev```, the output is:
```
libpmem-dev/bionic-updates,bionic-updates,now 1.4.1-0ubuntu1~18.04.1 amd64 [installed]
```

jim-wilson, 2021-06-06:
pkg-config returns true or false.  If using bash, you can do "pkg-config --exists libpmem; echo $?".  A 0 means success/true and a 1 means failure/false.

If you have libpmem-dev installed then you should have the /usr/include/libpmem.h file.
[https://packages.ubuntu.com/bionic-updates/libpmem-dev](url)
Under "list of files" it does include /usr/include/libpmem.h.  Maybe you should check to see if the file is there?  If not, maybe something is wrong with your system.  You would try reinstalling the package.

yslys, 2021-06-06:
Hi Jim, 

Thanks a lot for your guidance. I executed ```pkg-config --exists libpmem; echo $?``` and it returned 0.

I checked under the /usr/include directory, and the file libpmem.h is there; hence, I might need reinstall the package. Since I do not have the permission to run in sudo mode, I might have to wait until Monday or Tuesday to have the package reinstalled. Thanks again for your help.

jim-wilson, 2021-06-06:
If you do have the /usr/include/libpmem.h file, then why is the qemu build failing with an error saying that it can't find it?

yslys, 2021-06-07:
I am not sure why this error occurs. I have tried to build it twice, and failed for the same reason.

yslys, 2021-06-09:
The problem is solved by reinstalling the libpmem package.

closed 2021-06-09 by yslys

macromogic, 2021-07-18:
I have encountered the same problem on Ubuntu 20.04. I have the `/usr/include/libpmem.h` file, but BitBake still cannot find the header. I tried to append `/usr/include` to the include path, then I got even more compiler errors probably because of other header files in that directory.

My kernel version is 5.8.0-61-generic. Below is the result of `sudo apt list --installed | grep libpmem`:
```
libpmem-dev/focal,now 1.8-1ubuntu1 amd64 [installed]
libpmem1/focal,now 1.8-1ubuntu1 amd64 [installed]
libpmemblk-dev/focal,now 1.8-1ubuntu1 amd64 [installed]
libpmemblk1/focal,now 1.8-1ubuntu1 amd64 [installed]
libpmemlog-dev/focal,now 1.8-1ubuntu1 amd64 [installed]
libpmemlog1/focal,now 1.8-1ubuntu1 amd64 [installed]
libpmemobj-dev/focal,now 1.8-1ubuntu1 amd64 [installed]
libpmemobj1/focal,now 1.8-1ubuntu1 amd64 [installed]
libpmempool-dev/focal,now 1.8-1ubuntu1 amd64 [installed]
libpmempool1/focal,now 1.8-1ubuntu1 amd64 [installed]
```

davidlt, 2021-07-19:
You might try modifying https://github.com/openembedded/openembedded-core/blob/master/meta/recipes-devtools/qemu/qemu.inc

Add something like:
PACKAGECONFIG[pmem] = "--enable-libpmem,--disable-libpmem,libpmem"

macromogic, 2021-07-19:
Oh, it works. Thank you very much!