sifiveinc/freedom-u-sdk #151: Build clang failed due to incompatible `libz.so.1`

jwnhy, 2021-07-17:
When building SiFive SDK Coreip on Gentoo 5.10.27.

Task failed when compiling `clang`, I scan through the log find that it tells me it's due to incompatible `libz`.

```
| /home/john/tools/riscv-sifive/build/tmp-glibc/work/riscv64-oe-linux/clang/13.0.0-r0/recipe-sysroot-native/usr/bin/risc
v64-oe-linux/riscv64-oe-linux-ld: skipping incompatible /lib64/libz.so.1 when searching for /lib64/libz.so.1
| /home/john/tools/riscv-sifive/build/tmp-glibc/work/riscv64-oe-linux/clang/13.0.0-r0/recipe-sysroot-native/usr/bin/risc
v64-oe-linux/riscv64-oe-linux-ld: cannot find /lib64/libz.so.1
| /home/john/tools/riscv-sifive/build/tmp-glibc/work/riscv64-oe-linux/clang/13.0.0-r0/recipe-sysroot-native/usr/bin/risc
v64-oe-linux/riscv64-oe-linux-ld: skipping incompatible /lib64/libz.so.1 when searching for /lib64/libz.so.1
| clang-13: error: linker command failed with exit code 1 (use -v to see invocation)
| ninja: build stopped: subcommand failed.
```

My host `libz` is as follows. I have already compiled both 32/64 abi version of libz.
```
[ebuild   R    ] sys-libs/zlib-1.2.11-r4:0/1::gentoo  USE="minizip (split-usr) static-libs" ABI_X86="32 (64) (-x32)" 0 KiB
```

The complete build [log](https://pastebin.com/s46V0ZKg)

jim-wilson, 2021-07-19:
It is the build of the native riscv64-oe-linux clang that failed, which means it is the riscv64-oe-linux libz.so that has the problem.  There is a sysroot option
```--sysroot=/home/john/tools/riscv-sifive/build/tmp-glibc/work/riscv64-oe-linux/clang/13.0.0-r0/recipe-sysroot```
the broken libz.so is probably in there.  You would have to look at that to see what is wrong with it.

If the riscv64-oe-linux cross compiler is broken, then that could be due to an obscure bug in the native x86_64-linux compiler.  I don't know if gentoo is stable enough to build freedom-u-sdk.

davidlt, 2021-07-19:
Gentoo is not officially supported for OpenEmbedded project.

Does your target sysroot (`--sysroot=/home/john/tools/riscv-sifive/build/tmp-glibc/work/riscv64-oe-linux/clang/13.0.0-r0/recipe-sysroot`) has libz library?

jwnhy, 2021-07-20:
Thanks, guys, I built a docker to solve this issue...

closed 2021-07-20 by jwnhy