sifiveinc/freedom-u-sdk #153: Now in qemuriscv64_v mode, support to run exe with rv64(with V extension)?

vigoss30611, 2021-08-06:
When I run MACHINE=qemuriscv64_v runqemu nographic slirp, It show:
Boot HART ID              : 1
Boot HART Domain          : root
Boot HART ISA             : rv64imafdcvsu
Boot HART Features        : scounteren,mcounteren,time
Boot HART PMP Count       : 16
Boot HART PMP Granularity : 4
Boot HART PMP Address Bits: 54
Boot HART MHPM Count      : 0
Boot HART MHPM Count      : 0
Boot HART MIDELEG         : 0x0000000000000222
Boot HART MEDELEG         : 0x000000000000b109

It is seem support rv64imafdcvsu, but when I run exe with rv64(with V extension) in this qemu platform, 
show:
Program received signal SIGILL, Illegal instruction.
With the help of gdb, I find the reason for the illegal instruction with I run vsetivli isa...

jim-wilson, 2021-08-06:
Are you using the right version of the vector extension in your program?  There are many different incompatible draft versions of the vector spec, and if you don't have exactly the right draft version, then it won't work.  The qemu being used here only supports the 1.0 draft.  And clang is supporting the 0.10 draft which is close but not exactly the same.  The gcc here has no vector support.

IMHO I think the qemuriscv64_v target is only useful for internal SiFive testing, and won't be useable outside SiFive until there is less spec and toolchain churn, and more documentation available.  I'm not aware of any document I can point at that explains how to compile and run a RISC-V vector program.  Meanwhile, you are probably better off using a simpler environment, hand written rvv assembly code running on spike for instance.

vigoss30611, 2021-08-09:
> Are you using the right version of the vector extension in your program? There are many different incompatible draft versions of the vector spec, and if you don't have exactly the right draft [...]

I make the toolchain which support  rv64imafdv by myself. And It can run exe with the help of this toolchains, but it can not run vector extension(1.0 draft). For example vsetivli x0, 2, e64，I get assember content 0xc1817057  with the help of objdump,
unfortunately it show  Program received signal SIGILL, Illegal instruction when I run it

jim-wilson, 2021-08-09:
I suggest you debug qemu to see why it doesn't work.  And this is why I said it is only useful for internal SiFive testing, because users should not have to debug qemu to get something working.

davidlt, 2021-09-02:
This particular machine cannot yet run RVV application out-of-the box as mentioned in the README. This is just the first step to provide VM, but still there is no support on the run-time (e.g. kernel and glibc.). That will come later on.

closed 2021-09-02 by davidlt