sifiveinc/freedom-u-sdk #154: How to modify qemu source before build

yslys, 2021-08-30:
Hi all, 

I successfully built the SDK several months ago, and now I would like to configure my own device in qemu.

I started from the beginning, i.e. installing repo, and executing ```$ repo sync```, after which, there are five directories. I searched for some necessary information and found that the README's in [meta-sifive](https://github.com/sifive/meta-sifive/blob/2021.08/README.md) and [Sifive Freedom Unleashed SDK](https://github.com/sifive/freedom-u-sdk/blob/2021.08/README.md) are quite similar, which makes me confused:

- [meta-sifive](https://github.com/sifive/meta-sifive/blob/2021.08/README.md): SiFive OpenEmbedded Layer is a minimal meta-sifive layer on top of meta-riscv to provide additional modifications and new disk image targets.
- [Sifive Freedom Unleashed SDK](https://github.com/sifive/freedom-u-sdk/blob/2021.08/README.md): The new experimental Freedom Unleashed (FU) SDK is based on OpenEmbedded (OE). It's a minimal meta-sifive layer on top of meta-riscv to provide additional modifications and new disk image targets.

I am not sure what the difference is between meta-sifive and Sifive Freedom U SDK.

The five directories are listed below:
- [meta-clang](https://github.com/kraj/meta-clang): this layer provides clang/llvm as alternative to system C/C++ compiler for OpenEmbedded/Yocto Project based distributions.
- [meta-sifive](https://github.com/sifive/meta-sifive): a layer on top of meta-riscv to provide additional modifications and new disk image targets.
- [meta-riscv](https://github.com/riscv/meta-riscv): RISC-V Architecture Layer for OpenEmbedded/Yocto.
- [openembedded-core](https://github.com/openembedded/openembedded-core): a layer containing the core metadata for current versions of OpenEmbedded.
- [meta-openembedded](https://github.com/openembedded/openembedded-core): a layer containing the core metadata for current versions of OpenEmbedded.

Since I would like to modify qemu source before build, I tried to locate qemu directory by executing ```$ find . -name "qemu"``` in working directory. The result is as follows:
```
./openembedded-core/meta/recipes-devtools/qemu
./openembedded-core/meta/recipes-devtools/qemu/qemu
```

I went through such two directories, and I do not think those two are the source. I assume it is more likely that the files ending with .bb are the intermediate files that would be downloaded and built via bitbake (not quite sure if my assumption is correct).
 
Could anyone please tell me if there is any approach that I could modify qemu source before build? Thanks in advance.

jim-wilson, 2021-08-31:
freedom-u-sdk has gone through several major iterations.  In the current version of it, it is an OpenEmbedded layer, and as per OE conventions, the sources are in a repo called meta-sifive.  The freedom-u-sdk repo was retained for historical reasons, and now just has the release notes and the readme.  It also has old versions of the project on branches.

If you want to work with OpenEmbedded, you should read the OE docs.  There is a project called yocto that sits on top of OE that has a good manual.
[https://www.yoctoproject.org/docs/latest/mega-manual/mega-manual.html](url)

OE has rules to build packages from upstream sources.  OE specific changes are maintained as patch files, via the quilt program.  When you build a package, the sources get downloaded, the patches are applied, and then binaries are built.  So the sources will be found in the build tree.  If you want to modify them, then you need to add patch files into the source tree.

Current versions of freedom-u-sdk apply patches to qemu to add support for various extensions like V and B.  So qemu is actually built from meta-sifive/recipes-devtools/qemu now.  If you look at the qemu-sifive.inc file in that dir, you will see the list of patches to apply, and the order to apply them.  If you want to add your own patch, drop the patch file into the files dir, and then add the name of your patch file to the list in qemu-sifive.inc.

In my build tree, I can find qemu sources in build/tmp-glibc/work/x86_64-linux/qemu-native/6.0.0-r0/qemu-6.0.0.  Easiest way to work with this is to turn it into a git tree if it isn't already, and then use git to make patches, which you can then drop into the OE source tree, then rebuild to make sure the patch applies cleanly and works.

davidlt, 2021-09-02:
The way I like it to do is to use `-c devshell` with `bitbake`. That will fetch the sources and apply all the patches. I advice to use that with `tmux` and bitbake will detect that and drop you into a different pane. From here either create local git repo (`git init; git add *`) or one will already exist (if sources are fetched from git repository directly). Modify existing recipe file for QEMU and remove all the patches and point `SRC_URI` to your local git repository.

This way you can modify, commit, bitbake clean and build again.

For a very quick changes to test devshell is enough, i.e. in devshell you change code directly and then proceed to build. Another alternative would be to create a patch (e.g. in a devshell) and add into `SRC_URI` next to the other patches.

yuzibo, 2021-11-03:
As @jim-wilson  said, but detailed step are below:

```bash
# will generate workspace dir in build dir
devtool modify -x qemu-system-native  
# 2 fix your qemu code
# 3  rebuid qemu binaries
devtool build qemu-system-native 
# 4 re-constrcut your rootfs. The step confuses me also, 
# I do can not find a way rebuild qemu and its binaries will update immediately
# Until the step, you can fix qemu code.
MACHINE=qemuriscv64_b bitbake demo-coreip-cli
# 5 how to generate your qemu patch
git add/commit in workspace/source/qemu-xx/
# there is a default git repo, which branch is devtool-override
# 6 update recipes  in  oe-core layers's .bb files and generate patch
devtool update-recipe qemu-system-native
# 7 move oe-core's patch to meta-sifive's layers and update its bbappend file

```

yslys, 2022-01-09:
Thanks all for the useful comments.

I would like to provide another method to modify QEMU source **after** you have built SiFive Freedom-u-sdk.

By executing ```MACHINE=qemuriscv64 runqemu nographic slirp```, it is equivalent to executing a very long cmdline which will show up soon after you have executed ```MACHINE=qemuriscv64 runqemu nographic slirp```, right above the big OpenSBI logo. Something like this:
```
runqemu - INFO - Running .../build/tmp-glibc/work/x86_64-linux/qemu-helper-native/1.0-r1/recipe-sysroot-native/usr/bin/qemu-system-riscv64 -device ...
```
We could download the source of QEMU and compile it to RISCV architecture. Then, we could replace the first part (corresponding to qemu) of the above original command with the newly built qemu executable.

closed 2022-01-09 by yslys