sifiveinc/freedom-u-sdk #157: Repo unable to initialize/sync

DieracDelta, 2021-11-02:
Am I missing something? When I run:

```
repo init -u https://github.com/sifive/freedom-u-sdk.git -b 2021.09 -m tools/manifests
```
I get an error:
```
fatal: manifest 'manifests/sifive.xml' not available
fatal: manifest 'manifests/sifive.xml' not found
```

When I try with 2021.10 instead, initialization is successful, but `repo sync` fails:
```
fatal: remote error:
  The unauthenticated git protocol on port 9418 is no longer supported.
Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information.
```

My repo version is 2.17.3 (Wed, 27 Oct 2021 ...). Advice would be appreciated, thanks!

jim-wilson, 2021-11-02:
There was a restructuring between 2021.09 and 2021.10.  For 2021.09 you need to run repo init using sifive/meta-sifive.  With 2021.10 you need to run repo init using sifive/freedom-u-sdk.  The instructions in the README file for each release are correct.

OpenEmbedded is a large collection of random repos maintained by random people.  The github error message points at a URL that explains what changed, they deprecated some protocols and old ssh keys.  It will likely take OE a while to catch up to the change.  Meanwhile you can work around this by forcing use of a protocol that is still supported.  If I put this in my ~/.gitconfig file then repo sync works for me.
```
[url "ssh://git@github.com"]
        insteadOf = https://github.com
        insteadOf = http://github.com
        insteadOf = git://github.com
```
This assumes you have an ssh key uploaded to your github account.  Otherwise, you can try making https the primary protocol rather than ssh.

DieracDelta, 2021-11-03:
This worked! Thank you for the quick response.

closed 2021-11-03 by DieracDelta