sifiveinc/freedom-u-sdk #159: Error `trace/trace-ebpf.h: No such file or directory` when building qemu

yqszxx, 2021-11-07:
I got the following error message:
```
| In file included from ../qemu-6.1.0/ebpf/ebpf_rss.c:24:
| ../qemu-6.1.0/ebpf/trace.h:1:10: fatal error: trace/trace-ebpf.h: No such file or directory
|     1 | #include "trace/trace-ebpf.h"
|       |          ^~~~~~~~~~~~~~~~~~~~
| compilation terminated.
```
when building `qemu-native-6.1.0-r0` on Arch Linux.

According to [this](https://gitlab.com/qemu-project/qemu/-/issues/566) upstream issue, the workaround is to set `--disable-bpf` flag when configuring.

The diff is attached below:
```diff

project openembedded-core/
diff --git a/meta/recipes-devtools/qemu/qemu.inc b/meta/recipes-devtools/qemu/qemu.inc
index 57e5e1a015..29b4a24c22 100644
--- a/meta/recipes-devtools/qemu/qemu.inc
+++ b/meta/recipes-devtools/qemu/qemu.inc
@@ -94,6 +94,7 @@ EXTRA_OECONF = " \
     --with-git-submodules=ignore \
     --meson=meson \
     ${PACKAGECONFIG_CONFARGS} \
+    --disable-bpf \
     "
 ```

jim-wilson, 2021-11-07:
See also [https://github.com/sifive/meta-sifive/issues/33](https://github.com/sifive/meta-sifive/issues/33)

yqszxx, 2021-11-07:
Oh I didn’t see that! Let’s wait for upstream to fix it. Closing this.

closed 2021-11-07 by yqszxx