sifiveinc/freedom-u-sdk #164: Error: object name conflicts in archive

yslys, 2021-12-30:
Hi, I am building SiFive Freedom-u-sdk on Ubuntu 20.04.3, running on parallels desktop on ARM architecture (Apple M1 chip).

The command I executed was: ```PARALLEL_MAKE="-j 4" BB_NUMBER_THREADS=4 MACHINE=qemuriscv64 bitbake demo-coreip-cli```.

There was an error indicating that object name conflicts in archive. Below is the log file of failure plus the output before the log file which helps indicating the stage I was at. Note that the reason there was "Second Keyboard Interrupt, stopping..." at the end is because I hit Crtl+C to interrupt the task that was not finished since I built in parallel, so that the summary would be printed.

Thanks in advance!

```
Loading cache: 100% |##################################################################################################| Time: 0:00:00
Loaded 4102 entries from dependency cache.
Parsing recipes: 100% |################################################################################################| Time: 0:00:00
Parsing of 2698 .bb files complete (2697 cached, 1 parsed). 4103 targets, 209 skipped, 1 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.53.0"
BUILD_SYS            = "aarch64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "riscv64-oe-linux"
MACHINE              = "qemuriscv64"
DISTRO               = "nodistro"
DISTRO_VERSION       = "2021.12.00"
TUNE_FEATURES        = "riscv64"
meta                 = "work:88da9b61b469654805fd51869790b1fd6d34c5a3"
meta-oe              
meta-python          
meta-multimedia      
meta-filesystems     
meta-networking      
meta-gnome           
meta-xfce            = "work:2156c4416a711c42a9eda106ae228b53bd130ab7"
meta-riscv           = "work:9561639c61663a10d8c9c23d26173db499f4c39b"
meta-clang           = "work:fd587eaf0340faba2f85cf7b2e254a290500c48c"
meta-sifive          = "work:0eabb6030471da23789b4f0956e25767e966433d"
meta-tensorflow-lite = "work:1135f8bb469f1b7e36390c338dea9e62592daa49"
freedom-u-sdk        = "work:0d47c94368cbe5c5edaa1ba9bd0a7d504fc28b40"

Initialising tasks: 100% |#############################################################################################| Time: 0:00:01
Sstate summary: Wanted 1803 Local 0 Mirrors 0 Missed 1803 Current 1172 (0% match, 39% complete)
NOTE: Executing Tasks
ERROR: librsvg-native-2.52.4-r0 do_compile: oe_runmake failed
ERROR: librsvg-native-2.52.4-r0 do_compile: ExecutionError('/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/temp/run.do_compile.615715', 1, None, None)
ERROR: Logfile of failure stored in: /home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/temp/log.do_compile.615715
Log data follows:
| DEBUG: Executing python function autotools_aclocals
| DEBUG: SITE files ['endian-little', 'bit-64', 'arm-common', 'arm-64', 'common-linux', 'common-glibc', 'aarch64-linux', 'common']
| DEBUG: Python function autotools_aclocals finished
| DEBUG: Executing shell function do_compile
| NOTE: make -j 4
| make  all-recursive
| make[1]: Entering directory '/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/build'
| Making all in .
| make[2]: Entering directory '/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/build'
| /bin/bash ./libtool  --tag=CC   --mode=link gcc   -isystem/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/recipe-sysroot-native/usr/include -O2 -pipe -Wl,-Bsymbolic-functions -version-info 50:0:48 -export-dynamic -no-undefined -export-symbols-regex "^rsvg_.*"   -L/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/recipe-sysroot-native/usr/lib                         -L/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/yusen/riscv-sifive/build/tmp-glibc/sysroots-uninative/aarch64-linux/lib/ld-linux-aarch64.so.1 -o librsvg-2.la -rpath /home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/recipe-sysroot-native/usr/lib   librsvg_c_api.la -L/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/lib -lpng16 -lz -lcairo-gobject -lgdk_pixbuf-2.0 -lgio-2.0 -lxml2 -lpangocairo-1.0 -lcairo -lpangoft2-1.0 -lpango-1.0 -lgobject-2.0 -lglib-2.0 -lharfbuzz -lfontconfig -lfreetype -lm -ldl
| cd ../librsvg-2.52.4 &&							\
| PKG_CONFIG_ALLOW_CROSS=1						\
| PKG_CONFIG='/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/recipe-sysroot-native/usr/bin/pkg-config'						\
| CARGO_TARGET_DIR=/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/build/target			\
| cargo --locked build --verbose --target=aarch64-linux --release --bin rsvg-convert
| libtool: link: rm -fr  .libs/librsvg-2.exp .libs/librsvg-2.lax .libs/librsvg-2.so .libs/librsvg-2.so.2 .libs/librsvg-2.so.2.48.0 .libs/librsvg-2.ver
|        Fresh unicode-xid v0.2.2
|        Fresh autocfg v1.0.1
|        Fresh unicode-segmentation v1.8.0
|        Fresh pkg-config v0.3.22
|        Fresh strum v0.21.0
|        Fresh version-compare v0.0.11
|        Fresh ppv-lite86 v0.2.15
|        Fresh cfg-if v1.0.0
|        Fresh version_check v0.9.3
[... 202 lines cut ...]
| warning: `librsvg` (lib) generated 6 warnings
|        Fresh librsvg v2.52.4 (/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/librsvg-2.52.4)
|     Finished release [optimized] target(s) in 0.40s
| libtool: link: /home/yusen/riscv-sifive/build/tmp-glibc/hosttools/grep -E -e "^rsvg_.*" ".libs/librsvg-2.exp" > ".libs/librsvg-2.expT"
| libtool: link: mv -f ".libs/librsvg-2.expT" ".libs/librsvg-2.exp"
| libtool: link: echo "{ global:" > .libs/librsvg-2.ver
| libtool: link:  cat .libs/librsvg-2.exp | sed -e "s/\(.*\)/\1;/" >> .libs/librsvg-2.ver
| libtool: link:  echo "local: *; };" >> .libs/librsvg-2.ver
| libtool: link:  gcc  -shared  -fPIC -DPIC  -Wl,--whole-archive ./.libs/librsvg_c_api.a -Wl,--no-whole-archive  -L/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/recipe-sysroot-native/usr/lib -L/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/recipe-sysroot-native/lib -L/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/recipe-sysroot-native/usr/lib/pkgconfig/../../../usr/lib -lpng16 -lz -lcairo-gobject -lgdk_pixbuf-2.0 -lgio-2.0 -lxml2 -lpangocairo-1.0 -lcairo -lpangoft2-1.0 -lpango-1.0 -lgobject-2.0 -lglib-2.0 -lharfbuzz -lfontconfig -lfreetype -lm -ldl  -O2 -Wl,-Bsymbolic-functions -Wl,--enable-new-dtags -Wl,-rpath-link -Wl,/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/recipe-sysroot-native/usr/lib -Wl,-rpath-link -Wl,/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/recipe-sysroot-native/lib -Wl,-rpath -Wl,/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/recipe-sysroot-native/usr/lib -Wl,-rpath -Wl,/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/recipe-sysroot-native/lib -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/yusen/riscv-sifive/build/tmp-glibc/sysroots-uninative/aarch64-linux/lib/ld-linux-aarch64.so.1   -Wl,-soname -Wl,librsvg-2.so.2 -Wl,-version-script -Wl,.libs/librsvg-2.ver -o .libs/librsvg-2.so.2.48.0
| libtool: link: (cd ".libs" && rm -f "librsvg-2.so.2" && ln -s "librsvg-2.so.2.48.0" "librsvg-2.so.2")
| libtool: link: (cd ".libs" && rm -f "librsvg-2.so" && ln -s "librsvg-2.so.2.48.0" "librsvg-2.so")
[... cut ...]
| make[2]: *** [Makefile:814: librsvg-2.la] Error 1
| make[1]: *** [Makefile:1074: all-recursive] Error 1
| make[1]: Leaving directory '/home/yusen/riscv-sifive/build/tmp-glibc/work/aarch64-linux/librsvg-native/2.52.4-r0/build'
| make: *** [Makefile:707: all] Error 2

Second Keyboard Interrupt, stopping...

Summary: 1 task failed:
  virtual:native:/home/yusen/riscv-sifive/openembedded-core/meta/recipes-gnome/librsvg/librsvg_2.52.4.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

yslys, 2021-12-31:
@jim-wilson Thanks for your reply.

I am kind of confused since I did not have rust compiler installed prior to building this project. Following the link you provided, I installed rust and switched to nighly, but still did not solve the problem.

May I know what is the intuitive way to solve the version conflict issues? Thanks in advance!

jim-wilson, 2021-12-31:
I don't have a mac so I can't look into this.

jim-wilson, 2021-12-31:
I would look at the build log for the failing library to figure out what the file name conflict is, and then debug the build to figure out where the duplicate files are coming from, probably from two different places.  Maybe openembedded is building its own rust compiler, and that is the problem.  Or maybe I'm wrong about this being the same problem.

You don't have to build OE to use it.  You can download one of the prebuilt images from the releases section on the right.

If you do want to build OE, then you are probably better off using a linux machine.  The OE build process is complex, and MacOS is probably not a well supported build platform for OE.

yslys, 2022-01-01:
@jim-wilson Thanks a lot for your reply.

I am running Linux on a virtual machine on Mac, so it might not due to the non-linux machine problem.

I tried to reduce the problem to the smallest possible input that produces the bug: I downloaded the source of librsvg with version 2.52.5 from this [link](https://download-fallback.gnome.org/sources/librsvg/), and tried to install it using the following commands:
```
$ ./configure
$ sudo make install
```
I got the same error:
```
make  install-recursive
make[1]: Entering directory '/home/yusen/packages/librsvg-2.52.5'
Making install in .
make[2]: Entering directory '/home/yusen/packages/librsvg-2.52.5'
  CCLD     librsvg-2.la
libtool:   error: object name conflicts in archive: .libs/librsvg-2.lax/librsvg_c_api.a//home/yusen/packages/librsvg-2.52.5/./.libs/librsvg_c_api.a
make[2]: *** [Makefile:818: librsvg-2.la] Error 1
make[2]: Leaving directory '/home/yusen/packages/librsvg-2.52.5'
make[1]: *** [Makefile:1078: install-recursive] Error 1
make[1]: Leaving directory '/home/yusen/packages/librsvg-2.52.5'
make: *** [Makefile:1383: install] Error 2
```

According to what I got, I think what you mentioned is correct - openembedded is trying to build its own rust compiler, which is more likely to build librsvg as well.

yslys, 2022-01-01:
I noticed that there is an issue corresponding to this error: https://gitlab.gnome.org/GNOME/librsvg/-/issues/787

yslys, 2022-01-01:
For those who are using aarch64, one possible solution is to execute
```
repo init -u git://github.com/sifive/meta-sifive -b 2021.04 -m tools/manifests/sifive.xml
```
instead of
```
repo init -u git://github.com/sifive/freedom-u-sdk -b 2021.12 -m tools/manifests/sifive.xml
```
to build an older version of SiFive Freedom-u-sdk. 

This works for me on aarch64 Linux. Note that the version of QEMU is 5.2.0.

closed 2022-11-15 by tprrt