sifiveinc/freedom-u-sdk #165: How to build SiFive Freedom-u-sdk from scratch?

yslys, 2022-01-31:
### Background and Problem
Before using SiFive Freedom-u-sdk, I built a minimal riscv64 kernel on qemu following this [link](https://risc-v-getting-started-guide.readthedocs.io/en/latest/linux-qemu.html), but the **main issue** that I had was - binaries like ```git```, ```mknod```, ```ssh``` were not supported. So I switched to SiFive Freedom-u-sdk to rely on the convenience provided by it.

That would be great for those developers aiming to develop programs on SiFive board, but as a student, I would like to learn more. For instance, **how the disk image is built, i.e. what commands are actually executed so that it can support all those binaries**?

**I would like to re-produce the build process by typing in command lines without relying on the Yocto project.**

### What I did
I accidently caught the command used to emulate the system by checking the output immediately after executing ```$ MACHINE=qemuriscv64 runqemu nographic slirp```:

```
/nobackup/yusen/riscv-sifive/build/tmp-glibc/work/x86_64-linux/qemu-helper-native/1.0-r1/recipe-sysroot-native/usr/bin/qemu-system-riscv64 \
-device virtio-net-device,netdev=net0,mac=52:54:00:12:35:02 \
-netdev user,id=net0,hostfwd=tcp::2222-:22,hostfwd=tcp::2323-:23,tftp=/nobackup/yusen/riscv-sifive/build/tmp-glibc/deploy/images/qemuriscv64 \
-object rng-random,filename=/dev/urandom,id=rng0 \
-device virtio-rng-pci,rng=rng0 \
-drive id=disk0,file=/nobackup/yusen/riscv-sifive/build/tmp-glibc/deploy/images/qemuriscv64/demo-coreip-cli-qemuriscv64-20210701160127.rootfs.ext4,if=none,format=raw \
-device virtio-blk-device,drive=disk0 \
-device virtio-mouse-pci \
-device virtio-keyboard-pci \ 
-machine virt  -m 4096  \
-serial mon:stdio \
-serial null \
-nographic \
-device bochs-display \
-kernel /nobackup/yusen/riscv-sifive/build/tmp-glibc/deploy/images/qemuriscv64/Image--5.10.25+git0+031f6c76e4_cf5b0320cf-r0-qemuriscv64-20210701160127.bin \
-append 'root=/dev/vda rw  mem=4096M ip=dhcp console=ttyS0 console=hvc0 earlycon=sbi ' \
-bios /nobackup/yusen/riscv-sifive/build/tmp-glibc/deploy/images/qemuriscv64/fw_jump.elf
```
This allows me to test the functionality of various options. For instance, I have modified QEMU to simulate another device, and change the first line of the command to use my own QEMU to emulate the system. (Thanks for those who provide me with various solutions. For more information, see this [issue](https://github.com/sifive/freedom-u-sdk/issues/154)).

I have also compiled the kernel, and changed the ```-kernel``` option to emulate my own compiled kernel.

### Problem recap
But I got stuck when I was trying to figure out how the disk image and bios are built. May I know if there is a file that stores all the commands that executed during the build process? Or is there a way that can allow me to figure out what commands are executed?

jim-wilson, 2022-02-01:
I would suggest this
    [https://github.com/carlosedp/riscv-bringup/tree/master/Qemu](https://github.com/carlosedp/riscv-bringup/tree/master/Qemu)
There are also sister pages here that explain how to build images for the unleashed and unmatched boards.

Another option is to look at old freedom-u-sdk versions from before it switched to using OpenEmbedded.  The old versions just have a makefile to build everything, and all of the commands are in the makefile.  See for instance the freedom-u-sdk archive/buildroot branch.  But this hasn't been maintained in 3+ years so I don't know if it stills builds anymore.

yslys, 2022-02-02:
Thank you @jim-wilson! The information you provided is extremely helpful! I would close this issue.

closed 2022-02-02 by yslys