sifiveinc/freedom-u-sdk #174: freedom-u-sdk 2022.06 kas build error

JuFly, 2022-10-11:
Error while executing `kas build freedom-u-sdk/scripts/kas/qemuriscv64-cli.yml`.
`fatal: unable to access 'https://github.com/abseil/abseil-cpp/': Failed to connect to github.com port 443: Connection timed out`
[log.do_configure.2796418.txt](https://github.com/sifive/freedom-u-sdk/files/9750157/log.do_configure.2796418.txt)

But it is OK to execute `git clone https://github.com/abseil/abseil-cpp/`

tprrt, 2022-10-11:
Hi @JuFly,

I don't reproduce your issue, moreover as indicated in the issue #173 and log attached, it seems you are a connection issue to Github:
`
fatal: unable to access 'https://github.com/abseil/abseil-cpp/': Failed to connect to github.com port 443: Connection timed out
Cloning into 'abseil-cpp'...
fatal: unable to access 'https://github.com/abseil/abseil-cpp/': Failed to connect to github.com port 443: Connection timed out
Cloning into 'abseil-cpp'...
fatal: unable to access 'https://github.com/abseil/abseil-cpp/': Failed to connect to github.com port 443: Connection timed out
`

In addition, this issue occurs during the do_configure task, not from the do_fetch task, during the execution of cmake that tries to clone this git repository :
`  /data/htf/riscv-sifive/build/tmp/work/riscv64-freedomusdk-linux/libtensorflow-lite/2.9.0-r0/recipe-sysroot-native/usr/share/cmake-3.23/Modules/FetchContent.cmake:1217:EVAL:2 (__FetchContent_directPopulate)
 /data/htf/riscv-sifive/build/tmp/work/riscv64-freedomusdk-linux/libtensorflow-lite/2.9.0-r0/recipe-sysroot-native/usr/share/cmake-3.23/Modules/FetchContent.cmake:1217 (cmake_language)
`
Have you tried to do the same thing manually (without OE/yocto layers)?

_NB. Fetching additional sources through cmake isn't the right way, that can lead to a reproducible issue. All required sources should be cloned during the do_fetch task._

JuFly, 2022-10-11:
Thanks for your reply.
It's ok to execute `git clone https://github.com/abseil/abseil-cpp/` manually, so it might not be a connection issue.

tprrt, 2022-10-15:
Hi @JuFly,

Maybe from your shell environment do you have some variables to configure the network, for example, to define a proxy?

Can you try to execute the `kas build` command with in addition the argument `-E`?

Kind regards,
Thomas

JuFly, 2022-10-17:
Hi @tprrt
I removed the task about **libtensorflow-lite** according to `FAILED: /data/htf/riscv-sifive/build/tmp/work/riscv64-freedomusdk-linux/libtensorflow-lite/2.9.0-r0/build/src/abseil-cpp-populate-stamp/abseil-cpp-populate-download` and built the system normally.

I modified **IMAGE_INSTALL** in `/../../riscv-sifive/freedom-u-sdk/recipes-sifive/images/demo-coreip-cli.inc`.

closed 2022-10-17 by JuFly