sifiveinc/freedom-u-sdk #182: ZSTD symbol undeclared

yong-xuan, 2023-07-04:
Hi @nylon7,

The build system complains about the ZSTD symbol is undeclared.
```shell
| ../../git/src/compress.c: In function ‘uncompresszstd’:
| ../../git/src/compress.c:737:7: warning: implicit declaration of function ‘ZSTD_DCtx_reset’; did you mean ‘ZSTD_maxCLevel’? [-Wimplicit-function-declaration]
|   rc = ZSTD_DCtx_reset(zstd, ZSTD_reset_session_only);
|        ^~~~~~~~~~~~~~~
|        ZSTD_maxCLevel
| ../../git/src/compress.c:737:7: warning: nested extern declaration of ‘ZSTD_DCtx_reset’ [-Wnested-externs]
| ../../git/src/compress.c:737:29: error: ‘ZSTD_reset_session_only’ undeclared (first use in this function)
|   rc = ZSTD_DCtx_reset(zstd, ZSTD_reset_session_only);
|                              ^~~~~~~~~~~~~~~~~~~~~~~
| ../../git/src/compress.c:737:29: note: each undeclared identifier is reported only once for each function it appears in
| Makefile:575: recipe for target 'compress.lo' failed
| make[3]: *** [compress.lo] Error 1
| make[3]: *** Waiting for unfinished jobs....
| make[3]: Leaving directory '/scratch2/ywang2/test_fusdk/build/tmp/work/x86_64-linux/file-native/5.44-r0/build/src'
``` 

my commands are
```shell
kas shell --update ./freedom-u-sdk/scripts/kas/unmatched.yml
bitbake core-image-minimal
``` 

Thank you!

yong-xuan, 2023-07-04:
Hi Thomas,
I used the 2023.04 branch.

tprrt, 2023-07-04:
Hello @yong-xuan,

I didn't reproduce this issue, but it should be fixed in the 2023.06 by this [commit](https://git.openembedded.org/openembedded-core/commit/?id=b7c51fe740f00).

Kind regards,
Thomas

tprrt, 2023-07-04:
I closed this issue because it should be fixed since 2023.06.

closed 2023-07-04 by tprrt