sifiveinc/freedom-u-sdk #184: Building unmatched on 2023.06/2023.04 fails for rust

philippgie, 2023-07-12:
When running the build on 2023.04/2023.06 with  `kas build freedom-u-sdk/scripts/kas/unmatched.yml --target demo-coreip-cli` it fails continuously for rust and sometimes for bazel with 
```
|   cargo:warning=gcc: fatal error: cannot execute ‘cc1plus’: execvp: No such file or directory
```
for `/openembedded-core/meta/recipes-devtools/rust/rust_1.67.1.bb:do_compile`.

tprrt, 2023-07-13:
Hello @philippgie,

- Can you provide more information?
- Have you installed all the necessary packages on your distribution? (https://docs.yoctoproject.org/dev/singleindex.html#ubuntu-and-debian)

Kind regards,
Thomas

philippgie, 2023-07-13:
Hello @tprrt,

Thanks for looking into this!

What information could help?

I installed the necessary packages on my distribution which is Ubuntu 20.04 LTS.

```
gcc version 10.3.0 (Ubuntu 10.3.0-1ubuntu1~20.04) 
Python 3.8.10
tar (GNU tar) 1.30
git version 2.25.1
GNU Make 4.2.1
```

The error occurs on a clean clone of the repo and after running the command. The hardware also fulfills the requirements.

philippgie, 2023-07-24:
This seems to be an issue in my environment. Running in a docker container works flawlessly.