sifiveinc/freedom-u-sdk #190: Please make NVMe boot work out of the box

andreabolognani, 2025-02-26:
Despite the board having to built-in NVMe slot, it is possible to retrofit this functionality by purchasing an inexpensive adapter that fits into the PCIe slot.

NVMe support is compiled into U-Boot, so things mostly work already; however, a small tweak to the U-Boot environment is needed, specifically adding `nvme0` to `boot_targets` as such:

```
=> env default -f -a
## Resetting to default environment

=> env print boot_targets
boot_targets=mmc1 usb sata mmc0

=> env set boot_targets "mmc1 usb nvme0 sata mmc0"; env print boot_targets                    
boot_targets=mmc1 usb nvme0 sata mmc0

=> env save
Saving Environment to SPIFlash... Erasing SPI flash...Writing to SPI flash...done
OK
```

I have documented these steps on the [Fedora RISC-V wiki page dedicated to HiFive Premier P550](https://fedoraproject.org/wiki/Architectures/RISC-V/SiFive/HiFivePremierP550#NVMe_support_(optional)), but it would be fantastic if the default environment came with this change already built in, making manual user intervention unnecessary.

RHSI5, 2025-03-06:
We are exploring and testing NVMe boot along with SATA and will provide an update very soon.

andreabolognani, 2025-03-31:
@RHSI5 any update on this? IIRC a new firmware image for P550 was supposed to come out in March, I guess that slipped. I'd like to know what the current expected date is, and whether this change will make it in. Thanks!

RHSI5, 2025-03-31:
@andreabolognani , the release is ready and I expect it to be posted in a few days.

With this upcoming release I have tested booting the system from NVMe and USB drives (imaged on a host PC) as well as SATA and uSD card (imaged directly on Premier P550) and they all work. One just needs to ensure that only one boot device is connected to the system.

andreabolognani, 2025-03-31:
@RHSI5 that's great news, thank you!

To be clear, does that mean that the tweak suggested above (`boot_targets=mmc1 usb nvme0 sata mmc0`) has been included in the default environment, or are manual changes to the environment still necessary for the boot process to consider NVMe as a viable target?

RHSI5, 2025-04-01:
The defaults in the current build are:
boot_targets=ahci mmc1 usb nvme0 mmc0

During testing I noticed that the system selects /boot/efi rather randomly if multiple boot devices are present. One way to avoid this issue is to erase the eMMC in U-Boot therefore after flashing the external boot device erase the onboard eMMC using instructions in “HiFive Premier P550 Image Update Procedure” (https://www.sifive.com/document-file/hifive-premier-p550-image-update-procedure) section 6.1.

andreabolognani, 2025-04-01:
> The defaults in the current build are: boot_targets=ahci mmc1 usb nvme0 mmc0

I suppose `ahci` indicates the SATA disk?

The position of `nvme0` in the list makes sense, but shouldn't `ahci` come *after* `mmc1` and `usb`? You generally want removable storage to have precedence over internal storage. That's how things were configured in the previous build and it seemed to work very well - minus of course the lack of `nvme0` but thankfully you've addressed that ;)

It would IMO make a lot more sense to have

```
boot_targets="mmc1 usb ahci nvme0 mmc0"
```

instead, to maintain that desirable property.

> During testing I noticed that the system selects /boot/efi rather randomly if multiple boot devices are present. One way to avoid this issue is to erase the eMMC in U-Boot therefore after flashing [...]

It's been working very reliably for me, in the sense that when I have an SD card or an USB drive plugged in that *always* takes precedence over the internal NVMe drive, and the eMMC is simply never considered. Exactly as expected.

Have you by any chance flashed the same OS image to multiple devices? Because that will certainly cause things to work somewhat randomly. If Linux is configured to search the root filesystem by UUID, which is usually the case, and the same OS image with the same filesystem UUIDs has been written to multiple connected devices, there is simply no way for it to know which is which, and it will just pick one - possibly a different one every time. That shouldn't happen if the OS images are different.

andreabolognani, 2025-04-01:
Of course erasing the eMMC will work too and will, by definition, make things more predictable. I just don't think it should be necessary, and my experience so far seems to validate this belief.

jjardon, 2025-04-18:
I think this is now fixed as part of https://github.com/sifiveinc/freedom-u-sdk/releases/tag/2025.04.00-HFP550 ?

andreabolognani, 2025-04-22:
I have flashed the `2024.04.00-HFP550` release on my board and can confirm that NVMe boot now works out of the box, as does SATA boot.

Closing the issue now. Thanks a lot!