sifiveinc/freedom-u-sdk #191: HiFive Premier P550 firmware is outdated

andreabolognani, 2025-02-26:
The most recent build targeting the board is [`2024.11.00-HFP550`](https://github.com/sifive/freedom-u-sdk/releases/tag/2024.11.00-HFP550), which was tagged months ago and is marked as "pre-release".

According to the release notes, it comes with OpenSBI 1.4 and U-Boot 2024.01. Meanwhile, the most recent release for Unmatched is [`2025.02.00`](https://github.com/sifive/freedom-u-sdk/releases/tag/2025.02.00), which includes OpenSBI 1.6 and U-Boot 2025.01.

Please consider making a new firmware release targeting HiFive Premier P550 with updated OpenSBI and U-Boot. It would be great if #190 could be addressed as part of that release too.

RHSI5, 2025-03-06:
We will be releasing the next Yocto update for HiFive Premier P550 in March. The base U-Boot (2024.01) and OpenSBI (1.4) version are expected to stay the same. Most of the updates in the March release will be linux kernel related.

andreabolognani, 2025-03-06:
@RHSI5 that's a bit disappointing.

Can you please at least look into #190? It should be a trivial change on your part, but it would make the out of the box experience a lot better for those who want to use NVMe storage instead of SATA.